ciaa/firmware_v1 #189: Possible error in chain task

mcerdeiro, 2015-01-04:
OSEK-OS specification:
If the succeeding task is identical with the current task, this does not
result in multiple requests. The task is not transferred to the suspended
state, but will immediately become ready again.

y: 
When an extended task is transferred from suspended state into ready
state all its events are cleared.

por lo cual entiendo que una tarea extendida que hace chaintask los eventos no se deben setear a 0... tampoco hay un testcase para esto :(

gmuro, 2015-01-05:
¿Eso no se está haciendo ya en?:
https://github.com/ciaa/firmware.modules.rtos/blob/72f4d0e4b6f639a7b8be7bc75e08fc501420ca4f/src/ChainTask.c#L190

mcerdeiro, 2015-01-05:
tendrias razón si no fuera por que aca hacemos lo contrario antes:
https://github.com/ciaa/firmware.modules.rtos/blob/72f4d0e4b6f639a7b8be7bc75e08fc501420ca4f/src/ChainTask.c#L160

igual al parecer no hay testcase para esto, y la verdad no se que es mejor, me baje el osek erika para ver que hacen ellos y no entendi los fuentes heh :( son un lio... el ticket es para discutir que es mejor... :) no me queda claro

pd que bueno saber que alguien mas lee los issues :)

gmuro, 2015-01-05:
es cierto.
Entonces coincido en que los eventos no deben ser puestos a cero ya que la tarea no pasaría a suspended y luego a ready.

NataliaRequejo, 2015-03-27:
Por lo que entiendo si se llama a ChainTask y la tarea es idéntica a la que esta corriendo(= taskId), entonces la tarea no pasaría al estado SUSPENDED sino que seguiría en READY.
Solo deberían liberarse los recursos internos. Siguiendo esta linea de razonamiento en la linea https://github.com/ciaa/firmware.modules.rtos/blob/72f4d0e4b6f639a7b8be7bc75e08fc501420ca4f/src/ChainTask.c#L156
Agregaría la comparación  taskid != GetRunningTask()
 En caso que no lo fuera (idéntica) entonces sí pasaría al estado SUSPENDED y los eventos se resetearian a 0. 

Haciendo esa modificación, faltaría pensar si ejecutar o no el RemoveTask(GetRunningTask()); y el AddReady(). ya que se seguirá ejecutando la misma tarea y no va a cambiar de estado. Pero... puede que internamente este compartiendo un recurso y por lo tanto haya cambiado su prioridad segun OSEK Priority Ceiling Protocol. Por lo tanto dejaría esas dos llamadas tal cual están ahora.

mcerdeiro, 2015-03-27:
para mi hay que hacer un test case... :) y luego modificar el código...

Sacar RemoveTask y AddReady no se si es buena idea. Esto no se si esta especificado. Al hacer RemoveTask y AddReady estamos seguros que la tarea aparece última en la fifo. Si no hacemos esto un ChainTask es un while 1... Dejando esto la tarea se vuelve a agregar a la fifo, con lo cual si hay otras de la misma prioridad tambien se ejecutan. Por eso me parece mejor dejar RemoveTask y AddReady . Pero creo el standard ni los tests dicen nada al respecto.