ciaa/firmware_v1 #242: Issue While generate files

sharra7, 2015-04-17:
I am getting an issue while using "make generate" command .

php modules/rtos/generator/generator.php --cmdline -l -v \ -DARCH=x86 -DCPUTYPE=ia32 -DCPU=none \ -c examples/blinking/etc/blinking.oil -f ./modules/rtos/gen/inc/Os_Internal_Cfg.h.php ./modules/rtos/gen/inc/Os_Cfg.h.php ./modules/rtos/gen/src/Os_Cfg.c.php ./modules/rtos/gen/src/Os_Internal_Cfg.c.php ./modules/rtos/gen/src/x86/Os_Internal_Arch_Cfg.c.php ./modules/rtos/gen/inc/x86/Os_Internal_Arch_Cfg.h.php -o ./out/gen
ciaaFirmware RTOS Generator - Copyright 2008, 2009, 2015 Mariano Cerdeiro
Copyright 2014, ACSE & CADIEEL
ACSE : http://www.sase.com.ar/asociacion-civil-sistemas-embebidos/ciaa/
CADIEEL: http://www.cadieel.org.ar
All rights reserved.

INFO: the generator was called as follow:
INFO: modules/rtos/generator/generator.php --cmdline -l -v -DARCH=x86 -DCPUTYPE=ia32 -DCPU=none -c examples/blinking/etc/blinking.oil -f ./modules/rtos/gen/inc/Os_Internal_Cfg.h.php ./modules/rtos/gen/inc/Os_Cfg.h.php ./modules/rtos/gen/src/Os_Cfg.c.php ./modules/rtos/gen/src/Os_Internal_Cfg.c.php ./modules/rtos/gen/src/x86/Os_Internal_Arch_Cfg.c.php ./modules/rtos/gen/inc/x86/Os_Internal_Arch_Cfg.h.php -o ./out/gen 
INFO: ------ LICENSE START ------
INFO: This file is part of CIAA Firmware.
INFO: Redistribution and use in source and binary forms, with or without
INFO: modification, are permitted provided that the following conditions are met:
INFO: 1. Redistributions of source code must retain the above copyright notice,
INFO: this list of conditions and the following disclaimer.
INFO: 2. Redistributions in binary form must reproduce the above copyright notice,
INFO: this list of conditions and the following disclaimer in the documentation
INFO: and/or other materials provided with the distribution.
INFO: 3. Neither the name of the copyright holder nor the names of its
INFO: contributors may be used to endorse or promote products derived from this
INFO: software without specific prior written permission.
INFO: THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
INFO: AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
INFO: IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
INFO: ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE
INFO: LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
INFO: CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF
INFO: SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
INFO: INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
INFO: CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
INFO: ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
INFO: POSSIBILITY OF SUCH DAMAGE.
INFO: ------- LICENSE END -------
INFO: list of configuration files:
INFO: configuration file 1: examples/blinking/etc/blinking.oil
INFO: list of files to be generated:
INFO: generated file 1: ./modules/rtos/gen/inc/Os_Internal_Cfg.h.php
INFO: generated file 2: ./modules/rtos/gen/inc/Os_Cfg.h.php
INFO: generated file 3: ./modules/rtos/gen/src/Os_Cfg.c.php
INFO: generated file 4: ./modules/rtos/gen/src/Os_Internal_Cfg.c.php
INFO: generated file 5: ./modules/rtos/gen/src/x86/Os_Internal_Arch_Cfg.c.php
INFO: generated file 6: ./modules/rtos/gen/inc/x86/Os_Internal_Arch_Cfg.h.php
INFO: output directory: ./out/gen
INFO: reading examples/blinking/etc/blinking.oil

Notice: Undefined variable: config in C:\Rahul_Sharma\D_Drive\Rtos\Firmware-master\modules\rtos\generator\generator.php on line 334

Fatal error: Call to a member function parseOilFile() on null in C:\Rahul_Sharma\D_Drive\Rtos\Firmware-master\modules\rtos\generator\generator.php on line 334
Makefile:485: recipe for target `generate' failed

I am using cygwin on windows to build the project.

jcecconi, 2015-04-17:
Hi!
Please, do a 'git pull' first, and then update your makefile.mine based on
the new makefile.config
We are you from?
Bye
Juan

mcerdeiro, 2015-04-17:
I guess you are using the command line from Windows and not cygwin.

You have to install the IDE we also provide or manually. The problem is
that the manual is in Spanish. :(

Downloads
http://proyecto-ciaa.com.ar/devwiki/doku.php?id=descargas

IDE Docu
http://proyecto-ciaa.com.ar/devwiki/doku.php?id=docu:fw:bm:ide

You are going to need to use Google translate or similar tool..

Regards.
Mariano

mcerdeiro, 2015-04-17:
One more question, which repo are you using right now? The from github the release 0.4.1?

closed 2016-06-26 by mcerdeiro