ciaa/firmware_v1 #269: ¿Existen los tipos float estándar?

asCii88, 2015-06-04:
Hola, quería saber si debería existir float_t o cómo se recomienda trabajar si es sin ellos.

Muchas gracias.

mcerdeiro, 2015-06-04:
Hola,

a mi entender no tenes ese problema ya que float esta definido como 32 bits
y double de 64, en los enteros no es así. int esta definido como el tamano
del registro interno del procesador y al menos 16 bits. osea seguro son 16
bits pero pueden ser 32 o 64 segun la arq. osea tens 16,32 o 64. en cambio
en float y double no existe ese tema.

Buscando en el estándar de posix encuentro:
http://pubs.opengroup.org/onlinepubs/009695399/basedefs/math.h.html

pero no dice nada wow, osea float_t es al menos float. :) y double_t es al
menos double, pero usar float y double sigue sinedo claro y correcto.

en mi opinion creo usar float y double es correcto. tal vez es una buena
consulta para embebidos 32.

saludos.
Mariano.-

asCii88, 2015-06-04:
Muchas gracias

closed 2015-06-04 by asCii88

mcerdeiro, 2015-06-04:
De nada :)

gramoscelli, 2015-06-05:
Hola!
  En wikipedia se puede leer q la norma ieee 754 establece q es un valor de
punto flotante de 32 bits (año 1985). Mas tarde en el 2008 se modifico un
poco la representacion del valor en binario. En
http://en.wikipedia.org/wiki/IEEE_floating_point muestra que con  32 bits
se representa la presición simple y que la doble presición ocupa 64 bits.
En bajo nivel, mas alla de la arquitectura, se emula el coprocesador
matemático que traen los chips x86 desde el pentium i (alrededor de1990?).
Saludos!

Gustavo