ciaa/firmware_v1 #272: Interrupciones externas y ciaaDriverDio_ioctl

asCii88, 2015-06-05:
Hola, quería saber si se implementara la configuración de interrupciones externas debería hacerse utilizando la función del driver Dio. En ese caso qué literatura o ejemplos me recomendarían seguir para hacerlo de acuerdo al estándar y contribuir en algo.

Gracias.

mcerdeiro, 2015-06-05:
Hola,

Todavía no hay interrupciones de dio implementadas.

Que es lo que querés hacer? Y vemos si y como se podría implementar en el
firmware. Me parece va a ser complicado. Al menos siguiendo con posix.

Saludos
Mariano

asCii88, 2015-06-05:
Algo muy sencillo: configurar un par de pines para que generen una interrupción en flanco de bajada.

mcerdeiro, 2015-06-06:
Entiendo. La verdad no se bien. Pero ni es una issue. Asique dirá de
discutirlo en ciaa firmware. Estas en el grupo? La issue la cerraría.

Querés hacerlo en un pin cualquiera o en una entrada de la CIAA-NXP?

Saludos
Mariano

asCii88, 2015-06-06:
No sabía de ese grupo. Cierre nomás

mcerdeiro, 2015-06-06:
Listado de las listas :

http://proyecto-ciaa.com.ar/devwiki/doku.php?id=docu:intro

closed 2015-06-08 by asCii88