ciaa/firmware_v1 #285: DS en Makefile

cpantel, 2015-06-27:
Hay múltiples ocurrencias de "/" en lugar de $(DS):

371:	gcc $(addprefix $(OBJ_DIR)$(DS),$(UNITY_OBJ)) -lgcov -o out/bin/$(tst_file).bin
425:	ruby externals/ceedling/vendor/cmock/lib/cmock.rb -omodules/tools/ceedling/project.yml $(FILES_TO_MOCK)
519:	doxygen modules/tools/doxygen/doxygen.cnf
635:	@echo "   * the file ../Firmware.zip will be overwritten"
636:	@echo "   * the file ../Firmware.tar.gz will be overwritten"
641:	@echo Removing ../Firmware.zip
642:	rm -f ../Firmware.zip
643:	@echo Generating ../Firmware.zip
644:	zip -r ../Firmware.zip . -x *.git*
645:	@echo -f Removing ../Firmware.tar.gz
646:	tar -zcvf ../Firmware.tar.gz --exclude "*.git*" .
742:	@echo BOARD/ARCH/CPUTYPE/CPU...: $(BOARD)/$(ARCH)/$(CPUTYPE)/$(CPU)
803:	xsltproc /usr/share/xml/docbook/stylesheet/docbook-xsl/fo/docbook.xsl doc/documentation.xml > out/docbook/documentation.fo
804:	fop -fo out/docbook/documentation.fo -pdf out/docbook/documentation.pdf

mcerdeiro, 2015-06-27:
sip... :(

mcerdeiro, 2015-06-28:
lo único que no cambie fue la regla de docbook que en windows no puede correr, va en linux tampoco demaciado bien... nunca se uso, esta en standby el tema de documentación onda libro... otro punto abuerto del proyecto... si estas deacuerdo se puede cerrar la issue...

cpantel, 2015-06-28:
ok

closed 2015-06-28 by cpantel