ciaa/firmware_v1 #296: remove trailing spaces

mcerdeiro, 2015-08-02:


cpantel, 2015-08-06:
I propose to add a check in the Makefile in order to avoid committing this kind of errors

I would like to place all the logic in the Makefile, but I could not make it work, so this is my solution:
[Makefile]
code_sanity:
	./check_trailing_spaces.sh

osek_oil_gen_tst: code_sanity

some other test: code_sanity

(placing test_trailing_spaces.sh in some appropriate folder)

[check_trailing_spaces.sh]
TRAILING=$(find . -type f \
   -and -not -wholename "./.git*" \
   -and -not -wholename "./externals*" \
   -and -not -name "*.png" \
   -and -not -name "*~" \
   -and -not -name "*.zargo" \
   -and -not -name "*.axf" \
   -and -not -name "*.o" \
   -and -not -name "*.exe" \
   -and -not -name "*.a" \
   -and -not -name "*.swp" \
   -exec egrep -n -H " +$" {} \;)

if [ "$TRAILING"  ] ; then
   echo "FAIL: there are trailing spaces"
   echo "$TRAILING"
   exit 1
else
   echo "Trailing spaces OK"
fi

If you agree, I would try to extract some names from .gitignores instead of hardcoding

mcerdeiro, 2015-08-06:
I think is a great idea, the script can be stored in modules
https://github.com/ciaa/Firmware/tree/master/modules/tools/scripts

closed 2015-08-10 by mcerdeiro