ciaa/firmware_v1 #297: make me genera .exe en lugar de .axf

asCii88, 2015-08-03:
Hola, nuevamente no sé si preguntar esto por acá. Estoy intentando compilar en cygwin para la edu_ciaa. Cambié en el makefile.mine a BOARD ?= edu_ciaa_nxp.
Hago make clean, make generate y make -C /cygdrive/c/ciaa/firmware
Pero el archivo de salida que obtengo es un .exe
¿Hay que cambiar alguna otra cosa?

Por otra parte, cuando hago make openocd me sale "You can not start openocd in Windows nor Linux"

Copio la salida de la consola:

$ make -C /cygdrive/d/ciaa/firmware
make: se entra en el directorio '/cygdrive/d/ciaa/firmware'

===============================================================================
Compiling c file: ./modules/posix/src/ciaaSerialDevices.c

gcc -c -Wall -ggdb3 -c -Wall -ggdb3  -I`cygpath -w examples/adc_dac/inc`  -I./modules/posix/inc  -I./modules/ciaak/inc  -I./modules/drivers/inc  -I./modules/drivers/x86/inc  -I./modules/drivers/x86/ia32/inc  -I./modules/drivers/x86/ia32/none/inc  -I./externals/drivers/inc  -I./externals/drivers/x86/inc  -I./externals/drivers/x86/ia32/inc  -I./externals/drivers/x86/ia32/inc/device  -I./externals/drivers/x86/ia32/none/inc  -I./modules/rtos/inc  -I./modules/rtos/inc/x86  -I./modules/rtos/inc/x86/ia32  -I./out/gen/inc  -I./out/gen/inc/x86  -I./out/gen/inc/x86/ia32  -I./modules/libs/inc  -I./modules/base/inc  -I./modules/base/x86/inc  -I./modules/base/x86/ia32/inc  -I./modules/base/x86/ia32/none/inc  -I./externals/base/inc  -I./externals/base/x86/inc  -I./externals/base/x86/ia32/inc  -I./externals/base/x86/ia32/none/inc -DARCH=x86 -DCPUTYPE=ia32 -DCPU=none -DBOARD=edu_ciaa_nxp  `cygpath -w ./modules/posix/src/ciaaSerialDevices.c` -o ./out/obj/ciaaSerialDevices.o

Generating dependencies...
gcc -MM -c -Wall -ggdb3 -c -Wall -ggdb3  -I`cygpath -w examples/adc_dac/inc`  -I./modules/posix/inc  -I./modules/ciaak/inc  -I./modules/drivers/inc  -I./modules/drivers/x86/inc  -I./modules/drivers/x86/ia32/inc  -I./modules/drivers/x86/ia32/none/inc  -I./externals/drivers/inc  -I./externals/drivers/x86/inc  -I./externals/drivers/x86/ia32/inc  -I./externals/drivers/x86/ia32/inc/device  -I./externals/drivers/x86/ia32/none/inc  -I./modules/rtos/inc  -I./modules/rtos/inc/x86  -I./modules/rtos/inc/x86/ia32  -I./out/gen/inc  -I./out/gen/inc/x86  -I./out/gen/inc/x86/ia32  -I./modules/libs/inc  -I./modules/base/inc  -I./modules/base/x86/inc  -I./modules/base/x86/ia32/inc  -I./modules/base/x86/ia32/none/inc  -I./externals/base/inc  -I./externals/base/x86/inc  -I./externals/base/x86/ia32/inc  -I./externals/base/x86/ia32/none/inc -DARCH=x86 -DCPUTYPE=ia32 -DCPU=none -DBOARD=edu_ciaa_nxp  `cygpath -w ./modules/posix/src/ciaaSerialDevices.c` > ./out/obj/ciaaSerialDevices.d

===============================================================================
Compiling c file: ./modules/posix/src/ciaaDevices.c

gcc -c -Wall -ggdb3 -c -Wall -ggdb3  -I`cygpath -w examples/adc_dac/inc`  -I./modules/posix/inc  -I./modules/ciaak/inc  -I./modules/drivers/inc  -I./modules/drivers/x86/inc  -I./modules/drivers/x86/ia32/inc  -I./modules/drivers/x86/ia32/none/inc  -I./externals/drivers/inc  -I./externals/drivers/x86/inc  -I./externals/drivers/x86/ia32/inc  -I./externals/drivers/x86/ia32/inc/device  -I./externals/drivers/x86/ia32/none/inc  -I./modules/rtos/inc  -I./modules/rtos/inc/x86  -I./modules/rtos/inc/x86/ia32  -I./out/gen/inc  -I./out/gen/inc/x86  -I./out/gen/inc/x86/ia32  -I./modules/libs/inc  -I./modules/base/inc  -I./modules/base/x86/inc  -I./modules/base/x86/ia32/inc  -I./modules/base/x86/ia32/none/inc  -I./externals/base/inc  -I./externals/base/x86/inc  -I./externals/base/x86/ia32/inc  -I./externals/base/x86/ia32/none/inc -DARCH=x86 -DCPUTYPE=ia32 -DCPU=none -DBOARD=edu_ciaa_nxp  `cygpath -w ./modules/posix/src/ciaaDevices.c` -o ./out/obj/ciaaDevices.o

Generating dependencies...
gcc -MM -c -Wall -ggdb3 -c -Wall -ggdb3  -I`cygpath -w examples/adc_dac/inc`  -I./modules/posix/inc  -I./modules/ciaak/inc  -I./modules/drivers/inc  -I./modules/drivers/x86/inc  -I./modules/drivers/x86/ia32/inc  -I./modules/drivers/x86/ia32/none/inc  -I./externals/drivers/inc  -I./externals/drivers/x86/inc  -I./externals/drivers/x86/ia32/inc  -I./externals/drivers/x86/ia32/inc/device  -I./externals/drivers/x86/ia32/none/inc  -I./modules/rtos/inc  -I./modules/rtos/inc/x86  -I./modules/rtos/inc/x86/ia32  -I./out/gen/inc  -I./out/gen/inc/x86  -I./out/gen/inc/x86/ia32  -I./modules/libs/inc  -I./modules/base/inc  -I./modules/base/x86/inc  -I./modules/base/x86/ia32/inc  -I./modules/base/x86/ia32/none/inc  -I./externals/base/inc  -I./externals/base/x86/inc  -I./externals/base/x86/ia32/inc  -I./externals/base/x86/ia32/none/inc -DARCH=x86 -DCPUTYPE=ia32 -DCPU=none -DBOARD=edu_ciaa_nxp  `cygpath -w ./modules/posix/src/ciaaDevices.c` > ./out/obj/ciaaDevices.d

===============================================================================
Compiling c file: ./modules/posix/src/ciaaBlockDevices.c

gcc -c -Wall -ggdb3 -c -Wall -ggdb3  -I`cygpath -w examples/adc_dac/inc`  -I./modules/posix/inc  -I./modules/ciaak/inc  -I./modules/drivers/inc  -I./modules/drivers/x86/inc  -I./modules/drivers/x86/ia32/inc  -I./modules/drivers/x86/ia32/none/inc  -I./externals/drivers/inc  -I./externals/drivers/x86/inc  -I./externals/drivers/x86/ia32/inc  -I./externals/drivers/x86/ia32/inc/device  -I./externals/drivers/x86/ia32/none/inc  -I./modules/rtos/inc  -I./modules/rtos/inc/x86  -I./modules/rtos/inc/x86/ia32  -I./out/gen/inc  -I./out/gen/inc/x86  -I./out/gen/inc/x86/ia32  -I./modules/libs/inc  -I./modules/base/inc  -I./modules/base/x86/inc  -I./modules/base/x86/ia32/inc  -I./modules/base/x86/ia32/none/inc  -I./externals/base/inc  -I./externals/base/x86/inc  -I./externals/base/x86/ia32/inc  -I./externals/base/x86/ia32/none/inc -DARCH=x86 -DCPUTYPE=ia32 -DCPU=none -DBOARD=edu_ciaa_nxp  `cygpath -w ./modules/posix/src/ciaaBlockDevices.c` -o ./out/obj/ciaaBlockDevices.o

Generating dependencies...
gcc -MM -c -Wall -ggdb3 -c -Wall -ggdb3  -I`cygpath -w examples/adc_dac/inc`  -I./modules/posix/inc  -I./modules/ciaak/inc  -I./modules/drivers/inc  -I./modules/drivers/x86/inc  -I./modules/drivers/x86/ia32/inc  -I./modules/drivers/x86/ia32/none/inc  -I./externals/drivers/inc  -I./externals/drivers/x86/inc  -I./externals/drivers/x86/ia32/inc  -I./externals/drivers/x86/ia32/inc/device  -I./externals/drivers/x86/ia32/none/inc  -I./modules/rtos/inc  -I./modules/rtos/inc/x86  -I./modules/rtos/inc/x86/ia32  -I./out/gen/inc  -I./out/gen/inc/x86  -I./out/gen/inc/x86/ia32  -I./modules/libs/inc  -I./modules/base/inc  -I./modules/base/x86/inc  -I./modules/base/x86/ia32/inc  -I./modules/base/x86/ia32/none/inc  -I./externals/base/inc  -I./externals/base/x86/inc  -I./externals/base/x86/ia32/inc  -I./externals/base/x86/ia32/none/inc -DARCH=x86 -DCPUTYPE=ia32 -DCPU=none -DBOARD=edu_ciaa_nxp  `cygpath -w ./modules/posix/src/ciaaBlockDevices.c` > ./out/obj/ciaaBlockDevices.d

===============================================================================
Compiling c file: ./modules/posix/src/ciaaPOSIX_stdio.c

gcc -c -Wall -ggdb3 -c -Wall -ggdb3  -I`cygpath -w examples/adc_dac/inc`  -I./modules/posix/inc  -I./modules/ciaak/inc  -I./modules/drivers/inc  -I./modules/drivers/x86/inc  -I./modules/drivers/x86/ia32/inc  -I./modules/drivers/x86/ia32/none/inc  -I./externals/drivers/inc  -I./externals/drivers/x86/inc  -I./externals/drivers/x86/ia32/inc  -I./externals/drivers/x86/ia32/inc/device  -I./externals/drivers/x86/ia32/none/inc  -I./modules/rtos/inc  -I./modules/rtos/inc/x86  -I./modules/rtos/inc/x86/ia32  -I./out/gen/inc  -I./out/gen/inc/x86  -I./out/gen/inc/x86/ia32  -I./modules/libs/inc  -I./modules/base/inc  -I./modules/base/x86/inc  -I./modules/base/x86/ia32/inc  -I./modules/base/x86/ia32/none/inc  -I./externals/base/inc  -I./externals/base/x86/inc  -I./externals/base/x86/ia32/inc  -I./externals/base/x86/ia32/none/inc -DARCH=x86 -DCPUTYPE=ia32 -DCPU=none -DBOARD=edu_ciaa_nxp  `cygpath -w ./modules/posix/src/ciaaPOSIX_stdio.c` -o ./out/obj/ciaaPOSIX_stdio.o

Generating dependencies...
gcc -MM -c -Wall -ggdb3 -c -Wall -ggdb3  -I`cygpath -w examples/adc_dac/inc`  -I./modules/posix/inc  -I./modules/ciaak/inc  -I./modules/drivers/inc  -I./modules/drivers/x86/inc  -I./modules/drivers/x86/ia32/inc  -I./modules/drivers/x86/ia32/none/inc  -I./externals/drivers/inc  -I./externals/drivers/x86/inc  -I./externals/drivers/x86/ia32/inc  -I./externals/drivers/x86/ia32/inc/device  -I./externals/drivers/x86/ia32/none/inc  -I./modules/rtos/inc  -I./modules/rtos/inc/x86  -I./modules/rtos/inc/x86/ia32  -I./out/gen/i [... cut ...]
===============================================================================
Linking file: ./out/bin/adc_dac.exe

gcc ./out/obj/adc_dac.o -Xlinker --start-group  ./out/lib/posix.a  ./out/lib/ciaak.a  ./out/lib/drivers.a  ./out/lib/ext_drivers.a  ./out/lib/rtos.a  ./out/lib/libs.a -Xlinker --end-group -o ./out/bin/adc_dac.exe    -lpthread

===============================================================================
Post Building adc_dac

make: se sale del directorio '/cygdrive/d/ciaa/firmware'

jcecconi, 2015-08-03:
por favor, canalizar esto x la lista de CIAA-FW, no por github!!!

2015-08-03 11:13 GMT-03:00 asCii88 <notifications@github.com>:

> Hola, nuevamente no sé si preguntar esto por acá. Estoy intentando
> compilar en cygwin para la edu_ciaa. Cambié en el makefile.mine a BOARD ?=
> edu_ciaa_nxp.
> Hago make clean, make generate y make -C [...]

asCii88, 2015-08-03:
Hecho y cerrado.

closed 2015-08-03 by asCii88