ciaa/firmware_v1 #304: mismatch between ciaaCHANNEL_N and EDU-CIAA-NXP silkscreen

asCii88, 2015-08-06:
The ADC channel numbers do not correlate with the number written on the silkscreen. The current mapping is as follows:

ciaaCHANNEL_0 -> CH1
ciaaCHANNEL_1 -> CH2
ciaaCHANNEL_2 -> CH3
ciaaCHANNEL_3 -> NOTHING

The last thing also means that the example adc_dac is not functional https://github.com/ciaa/Firmware/blob/master/examples/adc_dac/src/adc_dac.c#L155

pridolfi, 2015-08-06:
This driver was designed for CIAA-NXP, not for EDU-CIAA-NXP. We need to improve it, like we did with the DIO driver.

pridolfi, 2015-08-06:
@asCii88 driver updated in new branch
https://github.com/ciaa/Firmware/tree/edu_ciaa_nxp_aio
can you test adc_dac example?

asCii88, 2015-08-06:
It is correct now. Every ciaaCHANNEL maps to its number.

Thanks.

closed 2015-08-10 by mcerdeiro