ciaa/firmware_v1 #335: Fix OSEK /rtos/tst !

sietedosfede, 2015-09-03:
Resuelto clonando https://github.com/ciaa/firmware.modules.rtos al directorio /Firmware/modules/rtos :
```
make: *** No rule to make target 'modules/rtos/tst/ctest/mak/Makefile'.  Alto. - blinking
233: modules/rtos/mak/Makefile: No existe el fichero o el directorio - blinking
842: modules/rtos/tst/ctest/mak/Makefile: No existe el fichero o el directorio - blinking
```
A resolver:
```
159: Symbol 'ActivatePeriodicTask' could not be resolved - blinking.c - /blinking/src
108: Symbol 'AppMode1' could not be resolved - blinking.c - /blinking/src
173: Type 'uint8_t' could not be resolved blinking.c - /blinking/src
```
[image]

mcerdeiro, 2015-09-03:
El primer tema es por que el clone no se hizo de la forma correcta. Se debe hacer con --recursive o hacer git submodules init git submodules updates. Esta en algún lado mal documentado? En el readme de https://github.com/ciaa/Firmware se esto claramente.

El otro tema no tiene nada que ver con el firmware. es un tema del ide. Al msmo hay que indicarle los paths correctos.

Esta queue es para reportar errores y no para soporte. para soporte pedimos se utilize el grupo: https://groups.google.com/forum/#!forum/ciaa-firmware

Para las dudas del eclipse pedimos usen el ide suite oficial o procedan a instalar como indicado en la pagina del proyecto:http://proyecto-ciaa.com.ar en caso de necesitar soporte para ide o hw o otros temas esta indicado:
http://proyecto-ciaa.com.ar/devwiki/doku.php?id=docu:intro

Saludos.
mariano.-

PD: el ticket como no es un error sera cerrado...:(

closed 2015-09-03 by mcerdeiro

sietedosfede, 2015-09-06:
Gracias por los links y la información útil!
Saludos,