ciaa/firmware_v1 #363: Not possible to have infinite loops in Autostart non cyclic Tasks?

ajithpadman, 2015-10-15:
I would like to know if it is possible to have Infinite While loop inside the Autostart tasks (which is not a cyclic task)which are supposed to be terminated after activating the other cyclic tasks? 
Currently when I tried it the other cyclic tasks not at all getting scheduled. Is there a way of doing this?

ajithpadman, 2015-10-15:
Or in more General Terms Is it possible to call the Scheduler explicitly without Terminating the Task?
I want the autostart non cyclic task to run as a background task.

mcerdeiro, 2015-10-15:
Hello Ajith,

we have an implementation of OSEK-VDX OS and not an AUTOSAR-OS.

 You can call the scheduler from a non preemptive task, the interface is
called Schedule.

From a full preemptive you dont have to do anything, the scheduler will
activate tasks with higher prio automatically.

So you can configure a low prio task full schedule and perform a while 1
and never end.

Personally I would not recommend such scheduling but it can be done. While
in while 1 the RTOS will never go to halt mode and consume a lot of current
all the time. You can never stop such task. Better is to have a ciclic task
or chain task at the end of a task to itself if needed.

Regards
Mariano

>
> Or in more General Terms Is it possible to call the Scheduler explicitly
without Terminating the Task?

closed 2015-10-15 by mcerdeiro