ciaa/firmware_v1 #382: Reference files for rtos osek OilGenerator functional testing drifted

cpantel, 2016-01-16:
The output from OilGenerator has changed and the files used as reference for the funcional testing are no longer valid. Created fix/ftest branch with improved error messages for the ftest

jcecconi, 2016-01-16:
Consulta..
Cuando un poil se modifica, el generador vuelve a generar los .oil?

cpantel, 2016-01-16:
No entiendo. ¿qué es un poil?
Lo que si entiendo es que los .oil no se generan, son definiciones de un proyecto, a los cuales OilGenerator usa para generar archivos .c y .h a partir de los templates *.c.php y *_h.php

jcecconi, 2016-01-16:
Los poil son oil con preprocesamiento, por ejemplo para sacar #ifdef, y
luego se generan los oil para el generate

cpantel, 2016-01-16:
ok, no hay .poil existentes aun, no? ¿Cuando decís "generan" querés decir "procesan"?

Con respecto a este issue, basta con comentar los "rm ..." en el test y ver si el _Cfg.h o _Cfg.c generado es correcto. De serlo, reemplazar la referencia de modo tal que en la próxima ejecución. Si me validás que son correctos, lo hago yo.

cpantel, 2016-01-16:
...tal que en la próxima ejecución [no falle]...

jcecconi, 2016-01-17:
Si correcto, me expresé mal. Los poil se procesan y a partir de ellos se
crean los oil. Acá hay algnos de ejemplo en el que estamos trabajando ahora

https://github.com/ciaa/Firmware/tree/feature/hisio/examples/hisio_example/etc

cpantel, 2016-01-17:
Ahi corregí los tests, sólo habría que verificar que las referencias en ./generator/tests/ftest/expected sean razonables

cpantel, 2016-05-01:
Entiendo que lo de los .poil no tienen nada que ver con este issue, cuando haga falta que alguien abra un nuevo issue con ese asunto, por favor

closed 2016-05-01 by cpantel