ciaa/firmware_v1 #414: Blinking example not working

evolentini, 2016-06-22:
After commit b39b07b of pull request #401 the blinking example does not compile, and after commit e418b10 this compile but not work. In a rapid debug session i found that never start the InitTask.

mabeett, 2016-06-22:
There was a module update also.
Being on `master` (`c5c69973e009094aa18bab8f58bfb6c9a685703f`) in the `make generate` step there are 3 warnings related with php.
`WARNING:    class 'Multicore' already exists, not loading again`

putting the module at 5e0cb73d64b58c16c7d21f9583161faf91994691 the blinking example will work fine. Not the blinking_lwip one.

```
cd modules/rtos
git checkout 5e0cb73d64b58c16c7d21f9583161faf91994691
```

mabeett, 2016-07-05:
Working at the current master (e418b10), the problem persist. 

With the Firmware repository at e418b10 and the submodule modules/rtos is at 5e0cb73 the blinking wiorks fine, but after f75f341 it does not work.

I am working on a solution studying the differences between sumbodule's commits. Work in progress.

closed 2016-07-06 by jcecconi