ciaa/firmware_v1 #417: ciaaPOSIX_usleep no toma valores menores a 1000000?

m4v, 2016-06-23:
En https://github.com/ciaa/Firmware/blob/master/modules/posix/inc/ciaaPOSIX_unistd.h#L86 dice
> \param[in] useconds micro seconds to sleep the application. Value shall be between 0 and 1000000.

valor correcto desde 0 a un millón, ok. Pero en https://github.com/ciaa/Firmware/blob/master/modules/posix/src/ciaaPOSIX_unistd.c#L146 tiene un check que retorna -1 si es menor a 1000000. Sospecho un error en el if, pero al mismo tiempo veo que los test unitarios en verifican justamente eso https://github.com/ciaa/Firmware/blob/master/modules/posix/test/utest/src/test_ciaaPOSIX_unistd.c#L443 algo menor a 1000000 debe retornar error.

Ya en este punto estoy completamente confundido sobre lo que se supone que usleep debe hacer.

mcerdeiro, 2016-06-24:
Suena razonable y parece ser un error. El finde miro tu propuesta y la
mergeo. :)

Saludos
Mariano

jcecconi, 2016-06-24:
Hay que ver bien el tema los utest cuando hay #if que modifican el código.
Entiendo que se debería hacer similar a como es en el rtos donde hay varios
test que definen lo necesario para probar todas las variantes de código.
Porque en este caso pasó el utest justamente porque el problema estaba
dentro de un #if no compilado

mcerdeiro, 2016-06-26:
Creo que podemos cerrar esta issue, no?

closed 2016-06-26 by mcerdeiro