ciaa/firmware_v1 #423: lwip + lpcopen: compiling for board differents than ciaa_nxp fails with no descriptive error message.

mabeett, 2016-07-01:
Since `externals/lwip/inc/arch/` and  `externals/lwip/src/arch/` are visible for any board when the user calls make for the `blinking_lwip` example the compiling fails getting `chip.h` headers.

`chip.h` file is only available for lpc43xx CPUTYPE

```
$ make 
 
===============================================================================
Compiling c file: ./externals/lwip/src/arch/lpc18xx_43xx_emac.c
 
gcc -c -Wall -ggdb3 -DCIAA_CFG_NET_IP -c -Wall -ggdb3  -Iexamples/blinking_lwip/inc  -I./modules/posix/inc  -I./modules/ciaak/inc  -I./modules/drivers/inc  -I./modules/drivers/x86/inc  -I./modules/drivers/x86/ia64/inc  -I./modules/drivers/x86/ia64/none/inc  -I./externals/drivers/inc  -I./externals/drivers/x86/inc  -I./externals/drivers/x86/ia64/inc  -I./externals/drivers/x86/ia64/inc/device  -I./externals/drivers/x86/ia64/none/inc  -I./modules/rtos/inc  -I./modules/rtos/inc/x86  -I./modules/rtos/inc/x86/ia64  -I./out/gen/inc  -I./out/gen/inc/x86  -I./out/gen/inc/x86/ia64  -I./modules/libs/inc  -I./modules/libs/inc/x86  -I./modules/libs/inc/x86/ia64  -I./externals/lwip/inc  -I./externals/lwip/inc/ipv4  -I./modules/base/inc  -I./modules/base/x86/inc  -I./modules/base/x86/ia64/inc  -I./modules/base/x86/ia64/none/inc  -I./externals/base/inc  -I./externals/base/x86/inc  -I./externals/base/x86/ia64/inc  -I./externals/base/x86/ia64/none/inc -DARCH=x86 -DCPUTYPE=ia64 -DCPU=none -DBOARD=ciaa_sim_ia64 ./externals/lwip/src/arch/lpc18xx_43xx_emac.c -o ./out/obj/lpc18xx_43xx_emac.o
In file included from ./externals/lwip/inc/lwip/arch.h:43:0,
                 from ./externals/lwip/inc/lwip/debug.h:35,
                 from ./externals/lwip/inc/lwip/opt.h:46,
                 from ./externals/lwip/src/arch/lpc18xx_43xx_emac.c:32:
./externals/lwip/inc/arch/cc.h:61:0: warning: "BYTE_ORDER" redefined [enabled by default]
 #define BYTE_ORDER LITTLE_ENDIAN
 ^
In file included from /usr/include/x86_64-linux-gnu/sys/types.h:216:0,
                 from ./modules/base/inc/ciaaPOSIX_stddef.h:77,
                 from ./modules/posix/inc/ciaaPOSIX_stdlib.h:51,
                 from examples/blinking_lwip/inc/lwipopts.h:119,
                 from ./externals/lwip/inc/lwip/opt.h:45,
                 from ./externals/lwip/src/arch/lpc18xx_43xx_emac.c:32:
/usr/include/endian.h:48:0: note: this is the location of the previous definition
 # define BYTE_ORDER __BYTE_ORDER
 ^
./externals/lwip/src/arch/lpc18xx_43xx_emac.c:45:18: fatal error: chip.h: No existe el archivo o el directorio
 #include "chip.h"
                  ^
compilation terminated.
make: *** [lpc18xx_43xx_emac.o] Error 1

$ make info| grep -i board  | head -n 1
BOARD/ARCH/CPUTYPE/CPU...: ciaa_sim_ia64/x86/ia64/none
$ find -name chip.h 
./out/download/lpcopen_2_16/lpc_chip_43xx/inc/chip.h
./out/download/lpcopen_2_16/lpc_chip_43xx_m0/inc/chip.h

```

mabeett, 2016-07-01:
Work in progress

mcerdeiro, 2016-07-01:
Lwip is only ported to nxp. Is not a bug but a feature. Nobody tried to
port Lwip to other board right now.

mcerdeiro, 2016-07-01:
Para mi hay que cerrar este issue y crearlo cuando alguien quiera ocuparse
del tema para otro board.

mabeett, 2016-07-01:
> Lwip is only ported to nxp. Is not a bug but a feature. Nobody tried to port Lwip to other board right now

`pc18xx_43xx_emac.c:45:18: fatal error: chip.h` does not seems being a feature related with te user. I am working in this ~~bug~~ _particular feature_.

> Para mi hay que cerrar este issue y crearlo cuando alguien quiera ocuparse del tema para otro board.

Me quiero ocupar para todos los boads. *Work in progress*. Estoy en eso. A la fecha saldría incompatinle con el otro MR (pero compatible con el actual master). Si el otro MR prospera antes, adecuaré los cambios.

mabeett, 2016-07-01:
Eh hecho de que esté trabajando en esto es para limpiar el directorio `arch` de los directorios de lwip,  entonces lograría llamar a lwip como submódulo del git repositoprio oficial de LWIP.
http://git.savannah.gnu.org/cgit/lwip.git/

Ya probé lwip 1.4.1 del upstream y funciona (no ipv6 aún). Probar la version 2.0.0 (beta) se trataría solamente de actualizar el módulo.

mabeett, 2016-10-07:
solved in #427 & #428

closed 2016-10-07 by mabeett