ciaa/firmware_v1 #432: example blinking_lwip not working

mabeett, 2016-07-06:
The blinking_lwip example does not work fine on ciaa_nxp. The code is build and downloaded normally. 
When the board runs the code the led blinks but there is no activity related with ethernet task. (no ping responses, no echo tcp). 

If the user sets `    ERRORHOOK = FALSE;` in the oil, the example works as usually.

```diff
diff --git a/examples/blinking_lwip/etc/blinking_lwip.oil b/examples/blinking_lwip/etc/blinking_lwip.oil
index 70eac73..10b801e 100644
--- a/examples/blinking_lwip/etc/blinking_lwip.oil
+++ b/examples/blinking_lwip/etc/blinking_lwip.oil
@@ -45,7 +45,7 @@ OSEK OSEK {
 
 OS ExampleOS {
     STATUS = EXTENDED;
-    ERRORHOOK = TRUE;
+    ERRORHOOK = FALSE;
     PRETASKHOOK = FALSE;
     POSTTASKHOOK = FALSE;
     STARTUPHOOK = FALSE;
```

mabeett, 2016-07-08:
Setting `ERRORHOOK = TRUE;` and deleting `ShutdownOS(0);` from `void ErrorHook()` the system seems working properly.
```diff
diff --git a/examples/blinking_lwip/src/blinking_lwip.c b/examples/blinking_lwip/src/blinking_lwip.c
index 37cca3e..d401bc8 100644
--- a/examples/blinking_lwip/src/blinking_lwip.c
+++ b/examples/blinking_lwip/src/blinking_lwip.c
@@ -100,7 +100,7 @@ void ErrorHook(void)
 {^M
    ciaaPOSIX_printf("ErrorHook was called\n");^M
    ciaaPOSIX_printf("Service: %d, P1: %d, P2: %d, P3: %d, RET: %d\n", OSErrorGetServiceId(), OSErrorGetParam1(), OSErrorGetParam2(), OSErrorGetParam3(), OSErrorGetRet());^M
-   ShutdownOS(0);^M
+//   ShutdownOS(0);^M
 }^M
 ^M
 TASK(InitTask)^M
```

mcerdeiro, 2016-07-08:
Hola Matías,

Con los cambios que hiciste no está funcionando bien. Esta funcionando mal
pero lo ignoras y por eso parece andar.

Deja el código como estaba. Pone un breakpoint en error hook y fijate los
valores de los parámetros que recibe error hook. También podes mirar el
stock trace..

Eso te va a dar una idea o me decís y yo te digo cual es el error.

ErrorHook va cuando hay una excepción y vos lo que hiciste fue de alguna
forma deshabilitar las excepciones. El problema sigue.

Se entiende?

Saludos
Mariano

<https://github.com/ciaa/Firmware/issues/432#issuecomment-231268586>, or mute
the thread

cpantel, 2016-07-08:
BTW, los ^M se pueden detectar con
https://github.com/ciaa/Firmware/blob/feature/coding_style/modules/tools/scripts/check_crlf.sh

y eliminar con 

https://github.com/ciaa/Firmware/blob/feature/coding_style/modules/tools/scripts/fix_crlf.sh
Saludos

 
-- Carlos Pantelides @dev4sec http://seguridad-agile.blogspot.com/

diff --git a/examples/blinking_lwip/src/blinking_lwip.c
b/examples/blinking_lwip/src/blinking_lwip.c
index 37cca3e..d401bc8 100644---
a/examples/blinking_lwip/src/blinking_lwip.c+++
b/examples/blinking_lwip/src/blinking_lwip.c@@ -100,7 +100,7 @@ void
ErrorHook(void)
{^M
ciaaPOSIX_printf("ErrorHook was called\n");^M
ciaaPOSIX_printf("Service: %d, P1: %d, P2: %d, P3: %d, RET: %d\n",
OSErrorGetServiceId(), OSErrorGetParam1(), OSErrorGetParam2(),
OSErrorGetParam3(), OSErrorGetRet());^M-  ShutdownOS(0);^M+//
ShutdownOS(0);^M
}^M
^M
TASK(InitTask)^M

mabeett, 2016-07-08:
Sí, eso veía, pero no llegué a escribirlo, me ganó el sueño. Me confundía que el ShutdownOS(0); no detiene a la placa (la tarea del parpadeo continua funcionando. Eso pasa porque el hook de shutodown está en false. No obstante poniendolo en true no está implemenatado el shutdown para la placa y falla en tiempo de compilación.
Lo que me indicás lo comprendo, voy a ver si llego a algo útil.

mabeett, 2016-07-08:
@cpantel entiendo, no estoy interesado en higienizar el código, los ^M vienen desde otros commits, el archivo está escrito a la manera de DOS y vim lo sigue escribiendo como lo leyó. Si cambio todo el archivo a UNIX no será visible lo que intervenga, es por eso que me parece mejor que un commit haga cambio de codificación y otro intervenga sobre la lógica del código. 

No obstante la comparación entre commits anteriores y posteriores de la higiene de código arrojará diferencias de este cambio de codificación más los cambios en la lógica, será (casi) imposible saber qué cambios en la lógica  hicieron entre esas dos versiones post y `pre fix_crlf.sh`

Si escribiera un nuevo archivo sería a la manera de UNIX, porque así lo hace vim en linux ;)

mabeett, 2016-07-08:
Apparently the was related with posix semaphores. 
[image]

When lwip calls `malloc()`, it `ciaaPOSIX_malloc()`is executed.

This patch may solve the problem.

```diff
--- a/examples/blinking_lwip/etc/blinking_lwip.oil
+++ b/examples/blinking_lwip/etc/blinking_lwip.oil
@@ -76,9 +77,12 @@ TASK InitTask {
 TASK PeriodicTask {
     PRIORITY = 1;
     ACTIVATION = 1;
     STACK = 2048;
     TYPE = BASIC;
     SCHEDULE = FULL;
+    RESOURCE = POSIXR;
 }

 TASK BlinkTask {
```

mabeett, 2016-07-10:
solved in #434

closed 2016-07-10 by mabeett