ciaa/firmware_v1 #437: INLINE defined and redefined during building

mabeett, 2016-07-13:
In the current master (`6ffe13f`) when building (for example) blinking example the compiler warnings advert about INLINE macro redefinitions.

```
In file included from ./modules/rtos/inc/Types.h:52:0,
                 from ./modules/rtos/inc/os.h:55,
                 from ./modules/drivers/cortexM4/lpc43xx/lpc4337/src/ciaaDriverAio.c:53:
./modules/rtos/inc/Compiler.h:81:0: warning: "INLINE" redefined
 #define INLINE
 ^
In file included from ./externals/drivers/cortexM4/lpc43xx/inc/cmsis.h:32:0,
                 from ./externals/drivers/cortexM4/lpc43xx/inc/chip.h:33,
                 from ./modules/drivers/cortexM4/lpc43xx/lpc4337/src/ciaaDriverAio.c:52:
./externals/drivers/cortexM4/lpc43xx/inc/lpc_types.h:205:0: note: this is the location of the previous definition
 #define INLINE inline
 ^
(...)
In file included from ./modules/rtos/inc/Types.h:52:0,
                 from ./modules/rtos/inc/os.h:55,
                 from ./modules/drivers/cortexM4/lpc43xx/lpc4337/src/ciaaDriverUart.c:53:
./modules/rtos/inc/Compiler.h:81:0: warning: "INLINE" redefined
 #define INLINE
 ^
In file included from ./externals/drivers/cortexM4/lpc43xx/inc/cmsis.h:32:0,
                 from ./externals/drivers/cortexM4/lpc43xx/inc/chip.h:33,
                 from ./modules/drivers/cortexM4/lpc43xx/lpc4337/src/ciaaDriverUart.c:52:
./externals/drivers/cortexM4/lpc43xx/inc/lpc_types.h:205:0: note: this is the location of the previous definition
 #define INLINE inline
 ^
 (...)
In file included from ./externals/drivers/cortexM4/lpc43xx/inc/cmsis.h:32:0,
                 from ./externals/drivers/cortexM4/lpc43xx/inc/chip.h:33,
                 from ./out/gen/src/cortexM4/Os_Internal_Arch_Cfg.c:59:
./externals/drivers/cortexM4/lpc43xx/inc/lpc_types.h:205:0: warning: "INLINE" redefined
 #define INLINE inline
 ^
In file included from ./modules/rtos/inc/os.h:55:0,
                 from ./modules/rtos/inc/Os_Internal.h:54,
                 from ./out/gen/src/cortexM4/Os_Internal_Arch_Cfg.c:54:
./modules/rtos/inc/Types.h:97:0: note: this is the location of the previous definition
 #define INLINE
 ^
```

mcerdeiro, 2016-07-13:
I think that the rtos does not use this macro so we can remove it. I have
to check it but I am almost sure about that. :)

jcecconi, 2016-07-13:
It is defined in  ["inc/Compiler.h" and "inc/Types.h"](https://github.com/ciaa/firmware.modules.rtos/search?utf8=%E2%9C%93&q=INLINE)  , maybe We can use a conditional #ifndef or remove it.

mabeett, 2016-07-30:
Sincerely I don´t know the context of the code causing the warnings.
If you want say me which surgery will be appropriate and I may elaborate and propose it as Pull Request.

I am searching for builds free of compiler warning messages, so the user might be able of pay attention to his own code warnings.

closed 2023-05-22 by mabeett