ciaa/firmware_v1 #444: Pointer to lseek is left uninitialized in ciaaSerialDevices_addDriver()

glpuga, 2016-09-28:
The following fragment of ciaaSerialDevices_addDriver() in ciaaSerialDevices.c should probably also copy the pointer for lseek.

```
/* allocate memory for new device */
newDevice = (ciaaDevices_deviceType*) ciaak_malloc(sizeof(ciaaDevices_deviceType));

/* set functions for this device */
newDevice->open = ciaaSerialDevices_open;
newDevice->close = ciaaSerialDevices_close;
newDevice->ioctl = ciaaSerialDevices_ioctl;
newDevice->read = ciaaSerialDevices_read;
newDevice->write = ciaaSerialDevices_write;

/* store layers information information */
newDevice->layer = (void *) &ciaaSerialDevices.devstr[position];
newDevice->loLayer = (void *) driver;
```

mabeett, 2016-09-29:
Related files in the current master branch
[modules/posix/src/ciaaSerialDevices.c#L157](https://github.com/ciaa/Firmware/blob/6ffe13f8986620f6853d5e407b02fa901d927534/modules/posix/src/ciaaSerialDevices.c#L157)
[modules/posix/inc/ciaaDevices.h#L107](https://github.com/ciaa/Firmware/blob/6ffe13f8986620f6853d5e407b02fa901d927534/modules/posix/inc/ciaaDevices.h#L107)

mcerdeiro, 2016-09-29:
Hello Gerardo,

This is a serial driver so it doesn't have a lseek. So I guess it shall
never been called. If so we have to set it to null.

What do you think?

glpuga, 2016-09-29:
Sounds ok to me.