ciaa/firmware_v1 #445: example blinking_multicore not working

mabeett, 2016-10-06:
A user cannot build blinking_multicore example with the information provided at repository

Reading the information in [cortexM0/blinking_multicore.c:107](../blob/6ffe13f8986620f6853d5e407b02fa901d927534/examples/blinking_multicore/src/cortexM0/blinking_multicore.c#L107) and [cortexM4/blinking_multicore.c:107](../blob/6ffe13f8986620f6853d5e407b02fa901d927534/examples/blinking_multicore/src/cortexM4/blinking_multicore.c#L107) there is no successful build.

I did not find a link or something similar for getting information.

```
developer@ciaa:/tmp/Firmware$ make info
Makefile:289: warning: overriding commands for target `out/lib/ext_drivers.a'
Makefile:289: warning: ignoring old commands for target `out/lib/ext_drivers.a'
+-----------------------------------------------------------------------------+
|               Enable Config Info                                            |
+-----------------------------------------------------------------------------+
Project Path.......: examples/blinking_multicore
Project Name.......: blinking_multicore
BOARD/ARCH/CPUTYPE/CPU...: ciaa_nxp/cortexM4/lpc43xx/lpc4337
enable modules.....: modules/posix modules/ciaak modules/drivers modules/rtos modules/libs modules/multicore modules/base
libraries..........: posix ciaak drivers ext_drivers rtos libs multicore ext_drivers base ext_base
libraris with srcs.: posix ciaak drivers ext_drivers rtos libs multicore ext_drivers ext_base
RTOS config........: examples/blinking_multicore/etc/blinking_multicore.oil
Includes...........: ./modules/posix/inc ./modules/ciaak/inc ./modules/drivers/inc ./modules/drivers/cortexM4/inc ./modules/drivers/cortexM4/lpc43xx/inc ./modules/drivers/cortexM4/lpc43xx/lpc4337/inc ./externals/drivers/inc ./externals/drivers/cortexM4/inc ./externals/drivers/cortexM4/lpc43xx/inc ./externals/drivers/cortexM4/lpc43xx/inc/device ./externals/drivers/cortexM4/lpc43xx/lpc4337/inc ./modules/rtos/inc ./modules/rtos/inc/cortexM4 ./modules/rtos/inc/cortexM4/lpc43xx ./out/gen/inc ./out/gen/inc/cortexM4 ./out/gen/inc/cortexM4/lpc43xx ./modules/libs/inc ./modules/libs/inc/cortexM4 ./modules/libs/inc/cortexM4/lpc43xx ./modules/multicore/inc ./modules/multicore/inc/cortexM4/lpc43xx ./externals/drivers/inc ./externals/drivers/cortexM4/inc ./externals/drivers/cortexM4/lpc43xx/inc ./externals/drivers/cortexM4/lpc43xx/inc/device ./externals/drivers/cortexM4/lpc43xx/lpc4337/inc ./modules/base/inc ./modules/base/cortexM4/inc ./modules/base/cortexM4/lpc43xx/inc ./modules/base/cortexM4/lpc43xx/lpc4337/inc ./externals/base/inc ./externals/base/cortexM4/inc ./externals/base/cortexM4/lpc43xx/inc ./externals/base/cortexM4/lpc43xx/lpc4337/inc
use make info_<mod>: to get information of a specific module. eg: make info_posix
+-----------------------------------------------------------------------------+
|               CIAA Firmware Info                                            |
+-----------------------------------------------------------------------------+
CIAA Firmware ver..: master-0.0.0
Available modules..: base ciaak drivers libs modbus multicore plc posix rtcs rtos systests template tools update
+-----------------------------------------------------------------------------+
|               Compiler Info                                                 |
+-----------------------------------------------------------------------------+
Compiler...........: gcc
CC.................: arm-none-eabi-gcc
AR.................: arm-none-eabi-ar
LD.................: arm-none-eabi-gcc
Compile C Flags....: -DMULTICORE -c -Wall -ggdb3 -fdata-sections -ffunction-sections -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -mthumb -DCORE_M4 -Iexamples/blinking_multicore/inc -I./modules/posix/inc -I./modules/ciaak/inc -I./modules/drivers/inc -I./modules/drivers/cortexM4/inc -I./modules/drivers/cortexM4/lpc43xx/inc -I./modules/drivers/cortexM4/lpc43xx/lpc4337/inc -I./externals/drivers/inc -I./externals/drivers/cortexM4/inc -I./externals/drivers/cortexM4/lpc43xx/inc -I./externals/drivers/cortexM4/lpc43xx/inc/device -I./externals/drivers/cortexM4/lpc43xx/lpc4337/inc -I./modules/rtos/inc -I./modules/rtos/inc/cortexM4 -I./modules/rtos/inc/cortexM4/lpc43xx -I./out/gen/inc -I./out/gen/inc/cortexM4 -I./out/gen/inc/cortexM4/lpc43xx -I./modules/libs/inc -I./modules/libs/inc/cortexM4 -I./modules/libs/inc/cortexM4/lpc43xx -I./modules/multicore/inc -I./modules/multicore/inc/cortexM4/lpc43xx -I./externals/drivers/inc -I./externals/drivers/cortexM4/inc -I./externals/drivers/cortexM4/lpc43xx/inc -I./externals/drivers/cortexM4/lpc43xx/inc/device -I./externals/drivers/cortexM4/lpc43xx/lpc4337/inc -I./modules/base/inc -I./modules/base/cortexM4/inc -I./modules/base/cortexM4/lpc43xx/inc -I./modules/base/cortexM4/lpc43xx/lpc4337/inc -I./externals/base/inc -I./externals/base/cortexM4/inc -I./externals/base/cortexM4/lpc43xx/inc -I./externals/base/cortexM4/lpc43xx/lpc4337/inc -DARCH=cortexM4 -DCPUTYPE=lpc43xx -DCPU=lpc4337 -DBOARD=ciaa_nxp
Compile ASM Flags..: -ggdb3 -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -mthumb
Target Name........: ./out/bin/blinking_multicore
Src Files..........: examples/blinking_multicore/src/cortexM4/blinking_multicore.c ./externals/base/cortexM4/lpc43xx/src/crp.c
Obj Files..........: blinking_multicore.o crp.o
Linker Flags.......: -fno-builtin -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -mthumb -Wl,-Map=./out/bin/blinking_multicore.map,-gc-sections -nostdlib -T externals/base/cortexM4/lpc43xx/linker/ciaa_lpc4337.ld
Linker Extension...: axf
Linker Target......: ./out/bin/blinking_multicore.axf
developer@ciaa:/tmp/Firmware$ make mcore 
Makefile:289: warning: overriding commands for target `out/lib/ext_drivers.a'
Makefile:289: warning: ignoring old commands for target `out/lib/ext_drivers.a'
***************************************
*** Making Cortex-M0 application... ***
***************************************
make ARCH=cortexM0 OUT_DIR=out/cortexM0
make[1]: Entering directory `/tmp/Firmware'
Makefile:289: warning: overriding commands for target `out/cortexM0/lib/ext_drivers.a'
Makefile:289: warning: ignoring old commands for target `out/cortexM0/lib/ext_drivers.a'
 
===============================================================================
Compiling c file: ./modules/posix/src/ciaaBlockDevices.c
 
arm-none-eabi-gcc -DMULTICORE -c -Wall -ggdb3 -fdata-sections -ffunction-sections -mcpu=cortex-m0 -mthumb -DCORE_M0 -D__MULTICORE_M0APP -DCORE_M0APP  -Iexamples/blinking_multicore/inc  -I./modules/posix/inc  -I./modules/ciaak/inc  -I./modules/drivers/inc  -I./modules/drivers/cortexM0/inc  -I./modules/drivers/cortexM0/lpc43xx/inc  -I./modules/drivers/cortexM0/lpc43xx/lpc4337/inc  -I./externals/drivers/inc  -I./externals/drivers/cortexM0/inc  -I./externals/drivers/cortexM0/lpc43xx/inc  -I./externals/drivers/cortexM0/lpc43xx/inc/device  -I./externals/drivers/cortexM0/lpc43xx/lpc4337/inc  -I./modules/rtos/inc  -I./modules/rtos/inc/cortexM0  -I./modules/rtos/inc/cortexM0/lpc43xx  -Iout/cortexM0/gen/inc  -Iout/cortexM0/gen/inc/cortexM0  -Iout/cortexM0/gen/inc/cortexM0/lpc43xx  -I./modules/libs/inc  -I./modules/libs/inc/cortexM0  -I./modules/libs/inc/cortexM0/lpc43xx  -I./modules/multicore/inc  -I./modules/multicore/inc/cortexM0/lpc43xx  -I./externals/drivers/inc  -I./externals/drivers/cortexM0/inc  -I./externals/drivers/cortexM0/lpc43xx/inc  -I./externals/drivers/cortexM0/lpc43xx/inc/device  -I./externals/drivers/cortexM0/lpc43xx/lpc4337/inc  -I./modules/base/inc  -I./modules/base/cortexM0/inc  -I./modules/base/cortexM0/lpc43xx/inc  -I./modules/base/cortexM0/lpc43xx/lpc4337/inc  -I./externals/base/inc  -I./externals/base/cortexM0/inc  -I./externals/base/cortexM0/lpc43xx/inc  -I./externals/base/cortexM0/lpc43xx/lpc4337/inc -DARCH=cortexM0 -DCPUTYPE=lpc43xx -DCPU=lpc4337 -DBOARD=ciaa_nxp ./modules/posix/src/ciaaBlockDevices.c -o out/cortexM0/obj/ciaaBlockDevices.o
In file included from ./modules/rtos/inc/os.h:57:0,
                 from ./modules/posix/src/ciaaBlockDevices.c:54:
./modules/rtos/inc/cortexM0/Os_Arch.h:56:34: fatal error: Os_Internal_Arch_Cfg.h: No such file or directory
 #include "Os_Internal_Arch_Cfg.h"
                                  ^
compilation terminated.
make[1]: *** [ciaaBlockDevices.o] Error 1
make[1]: Leaving directory `/tmp/Firmware'
make: *** [mcore] Error 2
developer@ciaa:/tmp/Firmware$ 
```

mcerdeiro, 2016-10-06:
I can confirm this issue :(

@cpantel could you check the generation, make mcore_generate seems not to be running
@pridolfi maybe you have an idea?

patriciobos, 2016-12-28:
I got this example back up and running by changing the following:

1. for the mcore_generate bug: 
Uncomment [line #88](https://github.com/ciaa/firmware.modules.rtos/blob/2c7034c0bc1c3b1c609ba774bb9170b14fb050e5/gen/src/cortexM0/Os_Internal_Arch_Cfg.c.php#L88) of Os_Internal_Arch_Cfg.c.php

2. For the build error that comes next...
Reduce the amount of reserved heap memory on [line#54](https://github.com/ciaa/Firmware/blob/master/modules/posix/src/ciaaPOSIX_stdlib.c#L54) of ciaaPOSIX_stdlib.c from 20000 to 12000

that worked for me, hope it rings any bell

pridolfi, 2016-12-28:
thanks for the workaround! please send us a pull-request mentioning this issue in order to fix it

fbucafusco, 2016-12-29:
Hi! For the MSP430 port I had to reduce the heap size as well and I implemented [this](https://github.com/fbucafusco/fw_on_msp430/blob/master/modules/posix/src/ciaaPOSIX_stdlib.c#L54) and [this](https://github.com/fbucafusco/fw_on_msp430/blob/master/examples/blinking/mak/Makefile#L63) for each project.

mabeett, 2017-04-14:
I have no opinion for the offered PR #454 . 
Is someone ready for accepting / rejecting it?

closed 2023-05-22 by mabeett