ciaa/firmware_v1 #447: NXPs: ciaaDriverDio.c differs for cortexM4 and cortexM4

mabeett, 2016-11-16:
In the current master (8a8e48f55cc4ffb8f6cb19e1688fe05496208ff5) the files `modules/drivers/cortexM4/lpc43xx/lpc4337/src/ciaaDriverDio.c` and `modules/drivers/cortexM0/lpc43xx/lpc4337/src/ciaaDriverDio.c` differs one from another.

The `ciaaDriverDio_dioType ciaaDriverDio_Outputs[]` are differents for edu-ciaa-nxp.

` ciaa_lpc4337_readPins()` and `ciaaDriverDio_write()` functions implementation differs one from another.

mabeett, 2016-11-16:
ciaa/Hardware@37110a6773f979839449b962433c38f5da111aa8 edu-ciaa-nxp does not use GPIO Ports  GPIO3[0], GPIO3[3] and GPIO3[4].
I saw my edu-ciaa-nxp and there are no vissible paths in this pins.

m4v, 2016-11-19:
> edu-ciaa-nxp does not use GPIO Ports GPIO3[0], GPIO3[3] and GPIO3[4].
I saw my edu-ciaa-nxp and there are no vissible paths in this pins

I believe I used those ports in the edu-ciaa-nxp, they are routed to the GPIO0, GPIO1, GPIO2 pins in the P2 connector.

mabeett, 2016-11-20:
> I believe I used those ports in the edu-ciaa-nxp, they are routed to the GPIO0, GPIO1, GPIO2 pins in the P2 connector.

I was absolutly wrong. I will close this issue and open one with only the observationgs regarding the differents files.

Thanks

closed 2016-11-20 by mabeett