ciaa/firmware_v1 #464: [WIP] ciaa_nxp: blinking_echo example does not build

mabeett, 2017-12-24:
blinkin_echo example, and probably others does not build.

```
developer@ciaa:/tmp/Firmware$ git  status ; git log --oneline -1 
On branch master
Your branch is up-to-date with 'origin/master'.

nothing to commit, working directory clean
7fb537a Merge pull request #463 from glpuga/features/portpicociaaleon3_merge_st4
developer@ciaa:/tmp/Firmware$ make info
+-----------------------------------------------------------------------------+
|               Enable Config Info                                            |
+-----------------------------------------------------------------------------+
Project Path.......: examples/blinking_echo
Project Name.......: blinking_echo
BOARD/ARCH/CPUTYPE/CPU...: ciaa_nxp/cortexM4/lpc43xx/lpc4337
enable modules.....: modules/posix modules/ciaak modules/drivers modules/rtos modules/libs modules/base
libraries..........: posix ciaak drivers ext_drivers rtos libs base ext_base
libraris with srcs.: posix ciaak drivers ext_drivers rtos libs ext_base
RTOS config........: examples/blinking_echo/etc/blinking_echo.poil
Includes...........: ./modules/posix/inc ./modules/ciaak/inc ./modules/drivers/inc ./modules/drivers/cortexM4/inc ./modules/drivers/cortexM4/lpc43xx/inc ./modules/drivers/cortexM4/lpc43xx/lpc4337/inc ./externals/drivers/inc ./externals/drivers/cortexM4/inc ./externals/drivers/cortexM4/lpc43xx/inc ./externals/drivers/cortexM4/lpc43xx/inc/device ./externals/drivers/cortexM4/lpc43xx/lpc4337/inc ./modules/rtos/inc ./modules/rtos/inc/cortexM4 ./modules/rtos/inc/cortexM4/lpc43xx ./out/gen/inc ./out/gen/inc/cortexM4 ./out/gen/inc/cortexM4/lpc43xx ./modules/libs/inc ./modules/libs/inc/cortexM4 ./modules/libs/inc/cortexM4/lpc43xx ./modules/base/inc ./modules/base/cortexM4/inc ./modules/base/cortexM4/lpc43xx/inc ./modules/base/cortexM4/lpc43xx/lpc4337/inc ./externals/base/inc ./externals/base/cortexM4/inc ./externals/base/cortexM4/lpc43xx/inc ./externals/base/cortexM4/lpc43xx/lpc4337/inc
use make info_<mod>: to get information of a specific module. eg: make info_posix
+-----------------------------------------------------------------------------+
|               CIAA Firmware Info                                            |
+-----------------------------------------------------------------------------+
CIAA Firmware ver..: master-0.0.0
Available modules..: base ciaak drivers libs modbus multicore plc posix rtcs rtos systests template tools update
+-----------------------------------------------------------------------------+
|               Compiler Info                                                 |
+-----------------------------------------------------------------------------+
Compiler...........: gcc
CC.................: arm-none-eabi-gcc
AR.................: arm-none-eabi-ar
LD.................: arm-none-eabi-gcc
Compile C Flags....: -c -Wall -ggdb3 -fdata-sections -ffunction-sections -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -mthumb -DCORE_M4 -Iexamples/blinking_echo/inc -I./modules/posix/inc -I./modules/ciaak/inc -I./modules/drivers/inc -I./modules/drivers/cortexM4/inc -I./modules/drivers/cortexM4/lpc43xx/inc -I./modules/drivers/cortexM4/lpc43xx/lpc4337/inc -I./externals/drivers/inc -I./externals/drivers/cortexM4/inc -I./externals/drivers/cortexM4/lpc43xx/inc -I./externals/drivers/cortexM4/lpc43xx/inc/device -I./externals/drivers/cortexM4/lpc43xx/lpc4337/inc -I./modules/rtos/inc -I./modules/rtos/inc/cortexM4 -I./modules/rtos/inc/cortexM4/lpc43xx -I./out/gen/inc -I./out/gen/inc/cortexM4 -I./out/gen/inc/cortexM4/lpc43xx -I./modules/libs/inc -I./modules/libs/inc/cortexM4 -I./modules/libs/inc/cortexM4/lpc43xx -I./modules/base/inc -I./modules/base/cortexM4/inc -I./modules/base/cortexM4/lpc43xx/inc -I./modules/base/cortexM4/lpc43xx/lpc4337/inc -I./externals/base/inc -I./externals/base/cortexM4/inc -I./externals/base/cortexM4/lpc43xx/inc -I./externals/base/cortexM4/lpc43xx/lpc4337/inc -DARCH=cortexM4 -DCPUTYPE=lpc43xx -DCPU=lpc4337 -DBOARD=ciaa_nxp
Compile ASM Flags..: -ggdb3 -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -mthumb
Target Name........: ./out/bin/blinking_echo
Src Files..........: examples/blinking_echo/src/blinking_echo.c ./externals/base/cortexM4/lpc43xx/src/crp.c
Obj Files..........: blinking_echo.o crp.o
Linker Flags.......: -fno-builtin -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -mthumb -Wl,-Map=./out/bin/blinking_echo.map,-gc-sections -nostdlib -T externals/base/cortexM4/lpc43xx/linker/ciaa_lpc4337.ld
Linker Extension...: axf
Linker Target......: ./out/bin/blinking_echo.axf

developer@ciaa:/tmp/Firmware$ make -s all 
Removing libraries
Removing bin files
Removing RTOS generated files
Removing mocks
Removing Unity Runners files
Removing doxygen files
Removing ci outputs
Removing coverage
Removing object files
Removing oil configuration files
Removing Download files
 
===============================================================================
Generate OIL File ./out/etc/blinking_echo.oil from examples/blinking_echo/etc/blinking_echo.poil
 
 
===============================================================================
Run RTOS Generator
 
INFO: the generator was called as follow:
INFO: modules/tools/generator/generator.php --cmdline -l -v -DARCH=cortexM4 -DCPUTYPE=lpc43xx -DCPU=lpc4337 -c ./out/etc/blinking_echo.oil -t ./modules/rtos/gen/inc/Os_Internal_Cfg.h.php ./modules/rtos/gen/inc/Os_Cfg.h.php ./modules/rtos/gen/src/Os_Cfg.c.php ./modules/rtos/gen/src/Os_Internal_Cfg.c.php ./modules/rtos/gen/src/cortexM4/Os_Internal_Arch_Cfg.c.php ./modules/rtos/gen/inc/cortexM4/Os_Internal_Arch_Cfg.h.php -o ./out/gen 
INFO: ------ LICENSE START ------
INFO: This file is part of CIAA Firmware.
INFO: Redistribution and use in source and binary forms, with or without
INFO: modification, are permitted provided that the following conditions are met:
INFO: 1. Redistributions of source code must retain the above copyright notice,
INFO: this list of conditions and the following disclaimer.
INFO: 2. Redistributions in binary form must reproduce the above copyright notice,
INFO: this list of conditions and the following disclaimer in the documentation
INFO: and/or other materials provided with the distribution.
INFO: 3. Neither the name of the copyright holder nor the names of its
INFO: contributors may be used to endorse or promote products derived from this
INFO: software without specific prior written permission.
INFO: THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
INFO: AND ANY EXPRESS OR IMP...
 
Generating dependencies...
 
===============================================================================
Linking file: ./out/bin/blinking_echo.axf
 
./out/lib/rtos.a(Os_Internal_Arch_Cfg.o): In function `Enable_User_ISRs':
/tmp/Firmware/./out/gen/src/cortexM4/Os_Internal_Arch_Cfg.c:209: undefined reference to `NVIC_EnableIRQ'
/tmp/Firmware/./out/gen/src/cortexM4/Os_Internal_Arch_Cfg.c:210: undefined reference to `NVIC_SetPriority'
/tmp/Firmware/./out/gen/src/cortexM4/Os_Internal_Arch_Cfg.c:213: undefined reference to `NVIC_EnableIRQ'
/tmp/Firmware/./out/gen/src/cortexM4/Os_Internal_Arch_Cfg.c:214: undefined reference to `NVIC_SetPriority'
/tmp/Firmware/./out/gen/src/cortexM4/Os_Internal_Arch_Cfg.c:217: undefined reference to `NVIC_EnableIRQ'
/tmp/Firmware/./out/gen/src/cortexM4/Os_Internal_Arch_Cfg.c:218: undefined reference to `NVIC_SetPriority'
collect2: error: ld returned 1 exit status
make[1]: *** [blinking_echo] Error 1
make: *** [all] Error 2
developer@ciaa:/tmp/Firmware$

mabeett, 2017-12-24:
PR open at rtos module repo: https://github.com/ciaa/firmware.modules.rtos/pull/41

glpuga, 2017-12-24:
The rtos submodule reference on the main repository was not updated after the latest batch of merges to the Firmeware.rtos repository. My fault, since I forgot to mention the need to do that when writing the merge request for the changes.

The rtos submodules in the current Firmware repository master needs to be updated to point to the current master of the Firmware.rtos repository. 

For the time being, stepping into the modules/rtos folder and executing "git checkout master" restores the build process. Can you confirm this works for you, @mabeett ?

mabeett, 2017-12-24:
> The rtos submodules in the current Firmware repository master needs to be updated to point to the current master of the Firmware.rtos repository.

> For the time being, stepping into the modules/rtos folder and executing "git checkout master" restores the build process. Can you confirm this works for you, @mabeett ?

Yes, checking out modules/rtos in master each example builds correctly[0]

[0] Disclaimer: I did no test the expected behavior for each example with the board.

glpuga, 2017-12-24:
@mabeett  If you agree, this issue should be renamed to "rtos submodules needs updating to current Firmware.rtos master." or something of that kind. I think only you can do that.

mabeett, 2017-12-24:
IMO, the issue should remain as is and  a new issue «rtos submodules needs updating to current Firmware.rtos master.» should be open.
When the new issue would be solved, this one should be closed too.
I mean this because the root cause es the submodule update, but this is not the issue itself.

Anyway, an administrator also may redefine the sbject an status of this issue.

mabeett, 2017-12-24:
mmm...
 Look at the current master, the problem persists for cortexM0 [Os_Internal_Arch_Cfg.c.php](
https://github.com/ciaa/firmware.modules.rtos/blob/52b3e757645a6d43e63fd196f3a652083dd388da/gen/src/cortexM0/Os_Internal_Arch_Cfg.c.php#L55)

glpuga, 2017-12-24:
> IMO, the issue should remain as is and a new issue «rtos submodules needs updating to current Firmware.rtos master.» should be open. When the new issue would be solved, this one should be closed too. I mean this because the root cause es the submodule update, but this is not the issue itself.

Ok, agree. Let's do that.

> mmm... Look at the current master, the problem persists for cortexM0 Os_Internal_Arch_Cfg.c.php

I've never compiled the firmware for the CortexM0 so I'm not sure what error are you seeing, but there is a mistake in the code line you linked: 

```
#include "Os_Internal.h"
#if (CPU == lpc43xx)
/* THIS IS A DIRTY WORKAROUND :( ciaa/Firmware#309*/
#undef FALSE
#undef TRUE
#include "chip.h"
#endif
```

lpc43xx is not a CPU, but a CPUTYPE. This is something was was corrected on the M4 code when merging the SPARC/picoCIAA code, but seems to have lingered on the M0 code. This code used to work because both "lpc43xx" and "lpc4337" had the same value, but I after the most recent platform additions that's no longer the case.

Try replacing "#if (CPU == lpc43xx)" with "#if (CPUTYPE == lpc43xx)" and see if that fixes the problem you are bumping into.

mabeett, 2017-12-24:
> Try replacing "#if (CPU == lpc43xx)" with "#if (CPUTYPE == lpc43xx)" and see if that fixes the problem you are bumping into.

Yes, this solves the problem.
I do not know in detail the system's file organization, bu I understand that the workaround should be applied in the same way for cotexM0 and cortexM4.
I mean `#if (CPUTYPE == lpc43xx)` or `#if (CPU == lpc4337)` in 
`gen/src/cortexM0/Os_Internal_Arch_Cfg.c.php` 
and
`gen/src/cortexM4/Os_Internal_Arch_Cfg.c.php`

glpuga, 2017-12-24:
> I do not know in detail the system's file organization, bu I understand that the workaround should be applied in the same way for cotexM0 and cortexM4. I mean #if (CPUTYPE == lpc43xx) or #if (CPU == [...]

This is already fixed for cortexM4 in the current master of Firmware.rtos. 

I'll create an issue for this.

closed 2023-05-22 by mabeett