ciaa/firmware_v1 #468: ISSUE WHILE GENERATING make

amirremad, 2022-12-26:
make generate is not working and give me this error !!
D:\OSEK_TRIAL\Firmware>make generate

===============================================================================
Run RTOS Generator

php modules/tools/generator/generator.php --cmdline -l -v \
        -DARCH=x86 -DCPUTYPE=ia32 -DCPU=none \
        -c  examples/blinking/etc/blinking.oil -t  ./modules/rtos/gen/inc/Os_Internal_Cfg.h.php  ./modules/rtos/gen/inc/Os_Cfg.h.php  ./modules/rtos/gen/src/Os_Cfg.c.php  ./modules/rtos/gen/src/Os_Internal_Cfg.c.php  ./modules/rtos/gen/src/x86/Os_Internal_Arch_Cfg.c.php  ./modules/rtos/gen/inc/x86/Os_Internal_Arch_Cfg.h.php -o ./out/gen
make: php: No such file or directory
make: *** [Makefile:601: gen.intermediate] Error 127

martinribelotta, 2022-12-26:
Need php binary in your path... this, or move to fw v3

amirremad, 2022-12-26:
How could i do the binary path ?
or can you give me the link of the new repo ?

amirremad, 2022-12-26:
i try the fw v3 it is doesnot work !
can you tell me how to add php binary in my path

cpantel, 2022-12-26:
Hi Amir

https://www.php.net/manual/en/install.windows.php

amirremad, 2022-12-26:
https://www.php.net/downloads
after I download this should I add the path of this file to the system environment variable?

cpantel, 2022-12-26:
I do not use windows, what I would do is:

https://devdojo.com/tnylea/installing-php-on-windows -> 3. Add PHP
Environment Variable

then, instead of powershell open a plain cmd to check if it is working

amirremad, 2022-12-26:
NOW I HAVE A NEW PROBLEM AFTER MAKING MAKE GENERATE 

Run RTOS Generator

php modules/tools/generator/generator.php --cmdline -l -v \
        -DARCH=x86 -DCPUTYPE=ia32 -DCPU=none \
        -c  examples/blinking/etc/blinking.oil -t  ./modules/rtos/gen/inc/Os_Internal_Cfg.h.php  ./modules/rtos/gen/inc/Os_Cfg.h.php  ./modules/rtos/gen/src/Os_Cfg.c.php  ./modules/rtos/gen/src/Os_Internal_Cfg.c.php  ./modules/rtos/gen/src/x86/Os_Internal_Arch_Cfg.c.php  ./modules/rtos/gen/inc/x86/Os_Internal_Arch_Cfg.h.php -o ./out/gen
PHP Fatal error:  Uncaught Error: Call to a member function printMsg() on null in D:\OSEK_TRIAL\Firmware\modules\tools\generator\OilGenerator.php:80
Stack trace:
#0 D:\OSEK_TRIAL\Firmware\modules\tools\generator\OilGenerator.php(146): OilGenerator->printCmdLine()
#1 D:\OSEK_TRIAL\Firmware\modules\tools\generator\OilGenerator.php(460): OilGenerator->processArgs(Array)
#2 D:\OSEK_TRIAL\Firmware\modules\tools\generator\generator.php(60): OilGenerator->run(Array)
#3 {main}
  thrown in D:\OSEK_TRIAL\Firmware\modules\tools\generator\OilGenerator.php on line 80

Fatal error: Uncaught Error: Call to a member function printMsg() on null in D:\OSEK_TRIAL\Firmware\modules\tools\generator\OilGenerator.php:80
Stack trace:
#0 D:\OSEK_TRIAL\Firmware\modules\tools\generator\OilGenerator.php(146): OilGenerator->printCmdLine()
#1 D:\OSEK_TRIAL\Firmware\modules\tools\generator\OilGenerator.php(460): OilGenerator->processArgs(Array)
#2 D:\OSEK_TRIAL\Firmware\modules\tools\generator\generator.php(60): OilGenerator->run(Array)
#3 {main}
  thrown in D:\OSEK_TRIAL\Firmware\modules\tools\generator\OilGenerator.php on line 80
make: *** [Makefile:601: gen.intermediate] Error 255

martinribelotta, 2022-12-27:
You requiere php v5 not more